isDipesh/gnome-shell-extension-sensors #124: Project has no explicit license

adrianbroher, 2015-06-09:
Hello everybody,

as you may know this project does not contain any license file or license statement.  While I don't know the legal implications of not having  any license, I would like to point out that the [extensions.gnome.org] page requires the extension maintainer to release the extension under the terms of the [GNU Public License version 2] or later.

Therefor I would like to ask every contributor to agree to change/apply the GPLv2 license to this project by stating his agreement this issue.

Why I'm asking specifically for the GPLv2 license you may ask; there are two reasons:

1. Not providing any GPLv2+ compatible license does not allow me to distribute this project on extensions.gnome.org
2. @UshakovVasilii provides a fork of this project with some great features but changed some modifications I don't like (like dropping localization support). He licensed his project under the terms of GPLv2 and applying that license to this project would allow some better exchange of features.

List of contributors (ordered as reported by the [contributors list]) agreeing to the licensing:

- [ ] @adrianbroher
- [ ] @xtranophilist 
- [ ] @farsx 
- [ ] @warmsun0220
- [ ] @UnsolvedCypher
- [ ] @Bouska
- [ ] @osmank3
- [ ] @perhenrik90
- [ ] @Tharre
- [ ] @ametheus
- [ ] @libook
- [ ] @lmacken
- [ ] @mackal
- [ ] @rgcjonas

[extensions.gnome.org]: https://extensions.gnome.org
[GNU Public License version 2]: https://www.gnu.org/licenses/gpl-2.0
[contributors list]: https://github.com/xtranophilist/gnome-shell-extension-sensors/graphs/contributors

adrianbroher, 2015-06-09:
For the sake of completeness:

I agree to licensing gnome-shell-extensions-sensors to the GPLv2+ license.

ametheus, 2015-06-09:
I, too, agree to licensing gnome-shell-extensions-sensors to the GPLv2+ license.

Tharre, 2015-06-09:
I agree to licensing gnome-shell-extensions-sensors to the GPLv2+ license.

Although, personally I'd prefer something like MIT or BSD for simplicity, both of which are GPL compatible.

farsx, 2015-06-09:
I agree to GPLv2+ for this extension

osmank3, 2015-06-09:
I agree to gpl v2+ for this extension

rgcjonas, 2015-06-10:
I agree to my contributions being licensed as GPLv2+, or, at your option, any other FSF-approved free software license.

lmacken, 2015-06-11:
I agree to my contributions being licensed as GPLv2+, or, at your option, any other FSF-approved free software license.

mackal, 2015-06-11:
I agree to my contributions being licensed as GPLv2+, or, at your option, any other FSF-approved free software license.

libook, 2015-06-11:
Sorry I am late for this.
Well I respect your choice, mate.
Many of my projects are using WTFPL, but they are just codes, designing means more things I think.
Thank you for designing this theme, which I've always wanted to say.

Wendor, 2015-07-02:
I agree to my contributions being licensed as GPLv2+, or, at your option, any other FSF-approved free software license.