isDipesh/gnome-shell-extension-sensors #129: Gnome 3.18

dmaglio, 2015-11-10:
Not compatible with Gnome 3.18 (Fedora 23)

mrgar, 2016-03-18:
Sadly same issue on Gnome 3.18.2. Extension installs fine but it cannot be activated. Warning message  says _'Extension does not support shell version'_

shusain, 2016-05-01:
If you edit ~/.local/share/gnome-shell/extensions/temperature@xtranophilist/manifest.json you can add 3.18 to the list of supported gnome-shell versions.  After that I restarted gnome (alt+f2 then type r and hit enter) and started the tweak tool and saw I could turn it on.

Enigma0, 2016-09-09:
Doesn't install for me.

failure84, 2017-02-02:
```
sudo apt install git lm-sensors gnome-common libglib2.0-dev libsensors-applet-plugin0
cd ~ && git clone https://github.com/xtranophilist/gnome-shell-extension-sensors.git
cd ~/gnome-shell-extension-sensors
./autogen.sh
make
make local-install
```

```
--- metadata.json.orig	2017-02-02 20:30:14.681381734 +0100
+++ metadata.json	2017-02-02 20:17:49.151644481 +0100
@@ -7,7 +7,8 @@
     "3.10",
     "3.12",
     "3.14",
-    "3.16"
+    "3.16",
+    "3.18"
   ],
   "localedir": "/usr/local/share/locale",
   "url": "https://github.com/xtranophilist/gnome-shell-extension-sensors", 
```
ALT+F2
r for reload
start tweak tool
enable the addon.

its works for me on this system:
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial
ii  gnome-shell             3.18.5-0ubuntu0. amd64            graphical shell for the GNOME desktop