isDipesh/gnome-shell-extension-sensors #136: Does not work. TypeError: PanelMenu.SystemStatusButton is undefined

ashirviskas, 2017-05-11:
GNOME Shell 3.24.1
Ubuntu Gnome 17.04

I managed to get it working somehow (installing and reinstallng), but after a reboot it does not work anymore.

Any more info needed?

qakcn, 2017-05-12:
Using Ubuntu Gnome 17.04 too.

For me, the erroe message is **TypeError: Main.messageTray._messageTrayMenuButton is undefined**

MisterGuinness, 2017-06-25:
I received this error on Fedora 25 after migrating from Fedora 22 (where I've had this extension working for years). Initially I installed the extension from the Software app. When I compared the source code to what is on git, the version installed as a very old one.

To fix, remove the old extension then follow the manual installation instructions in the README on the Code tab, to clone the git source, etc. Under wayland alt+F2 and "r" does not work, so just logout and back in to restart gnome shell.  Turn on the extension using gnome-tweak-tool.

Note you need to install lm_sensors (and optionally hddtemp) and run sensors-detect to configure.