isDipesh/gnome-shell-extension-sensors #137: Problems with GNOME extension sensor

thedoctor818, 2017-05-24:
I am trying to install the sensor gnome extension and all I see in the top panel is: "Error". Upon clicking on that, I see the following: "Please install lm_sensors. If that doesn't help click here to report with your sensor output." I just attempted to install said software via the ubuntu repositories (am running Ubuntu 17.04 on this machine) and no go still getting the same message. I am not sure what to do as I am totally new to extensions - trying them out as I switch from Unity to Gnome in advance (as many people probably are) of Canonical dumping Unity. Thanks, in advance, for any help you can be.

mwhankins, 2017-06-23:
Try typing the following in a terminal.
sudo apt-get install lm-sensors

MisterGuinness, 2017-06-25:
Also don't forget to run "sensors-detect" in a terminal to configure the sensors for lm_sensors. You can test by then running "sensors" to get the output for all the configured sensors. The extension is working for me on Fedora 25, but had to use the manual installation instructions in the README file.