isDipesh/gnome-shell-extension-sensors #142: Not working on gnome 3.32.1

shashank2806, 2019-05-05:
There is error loading extension
```
SyntaxError: missing ( after for

Stack trace:
  _getExtensionPrefsModule@resource:///org/gnome/shell/extensionPrefs/main.js:59:13
  _selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:74:31
  _onCommandLine@resource:///org/gnome/shell/extensionPrefs/main.js:320:17
  main@resource:///org/gnome/shell/extensionPrefs/main.js:635:5
  @<main>:1:43
```

EthanDarling8, 2019-05-17:
I'm getting the same problem. Running Ubuntu 19.04 it gives me this:

The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```
Stack trace:
```
_getExtensionPrefsModule@resource:///org/gnome/shell/extensionPrefs/main.js:59:13
_selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:74:31
_onCommandLine@resource:///org/gnome/shell/extensionPrefs/main.js:320:17
main@resource:///org/gnome/shell/extensionPrefs/main.js:635:5
@<main>:1:43
```
`

Eudj, 2019-06-27:
same thing

The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_getExtensionPrefsModule@resource:///org/gnome/shell/extensionPrefs/main.js:59:13
_selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:74:31
_extensionFound/<@resource:///org/gnome/shell/extensionPrefs/main.js:278:13
main@resource:///org/gnome/shell/extensionPrefs/main.js:635:5
@<main>:1:43
```