isDipesh/gnome-shell-extension-sensors #143: Not working on Ubuntu 19.04

amalic, 2019-05-31:
I installed lm-sensors as described and did sensor-detect as root. When enabling the extension it says Error. When clicking on the config icon i get following error messge:
```
SyntaxError: missing ( after for

Stack trace:
  _getExtensionPrefsModule@resource:///org/gnome/shell/extensionPrefs/main.js:59:13
  _selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:74:31
  _onCommandLine@resource:///org/gnome/shell/extensionPrefs/main.js:320:17
  main@resource:///org/gnome/shell/extensionPrefs/main.js:635:5
  @<main>:1:43
```

UshakovVasilii, 2019-06-01:
Possible master not work with old gnome-shell version (`gnome-shell --version`). You can install correct version from https://extensions.gnome.org/extension/841/freon/

petermat, 2019-07-28:
same here, fresh install 19.04