isDipesh/gnome-shell-extension-sensors #148: Error installing

ManuelAntonioKuhn, 2020-03-27:
The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
get prefsModule@resource:///org/gnome/shell/extensionPrefs/main.js:675:13
_showPrefs@resource:///org/gnome/shell/extensionPrefs/main.js:57:13
_extensionsLoaded@resource:///org/gnome/shell/extensionPrefs/main.js:320:13
_scanExtensions/<@resource:///org/gnome/shell/extensionPrefs/main.js:298:13
asyncCallback@resource:///org/gnome/gjs/modules/overrides/Gio.js:132:13
main@resource:///org/gnome/shell/extensionPrefs/main.js:707:5
@<main>:1:43
```
[image]

Yaya0312, 2020-04-23:
I have the same issue

SyntaxError: 

```
missing ( after for
```

Stack trace:
  ```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:222:19
  start@resource:///org/gnome/gjs/modules/script/package.js:206:5
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17

```

[image]

Alex815O, 2020-04-30:
There is the same issue there with a solution: https://github.com/xtranophilist/gnome-shell-extension-sensors/issues/150

alireza1371, 2021-03-06:
> +1

+1