isDipesh/gnome-shell-extension-sensors #150: I recieved this error - ubuntu 20.04lts - missing  " ( "

bettencb, 2020-04-29:
The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

Alex815O, 2020-04-30:
I had the same issue. There are syntax errors in three different files. 
Those files are in the directory:
>  ~/.local/share/gnome-shell/extensions/temperature@xtranophilist/
Files: 
- extension.js
- convenience.js
- utilities.js

You can open them with a text editor and replace every "for each" to "for".

closed 2020-05-28 by bettencb

danielsanfr, 2020-05-28:
Why did you close PR? Is there any solution?

bettencb, 2020-05-29:
yes, read above

plibither8, 2020-08-15:
Even after fixing the instances of "for each" to "for", how do I reinitiate that extension, so it uses the changed code? This doesn't seem to be a solid fix :/