isDipesh/gnome-shell-extension-sensors #151: syntax error

lsatenstein, 2020-06-12:
SyntaxError: missing ( after for

Stack trace:
  _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:222:19
  start@resource:///org/gnome/gjs/modules/script/package.js:206:5
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17

c0dehard, 2020-06-14:
### Almost same here using wayland gnome @ Linux 4.19.108-1 MANJARO-ARM 20.04 aarch64

The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
get prefsModule@resource:///org/gnome/shell/extensionPrefs/main.js:732:13
_showPrefs@resource:///org/gnome/shell/extensionPrefs/main.js:174:13
_extensionsLoaded@resource:///org/gnome/shell/extensionPrefs/main.js:467:18
_scanExtensions/<@resource:///org/gnome/shell/extensionPrefs/main.js:418:18
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
main@resource:///org/gnome/shell/extensionPrefs/main.js:761:23
@<main>:1:48
```

MrMic, 2020-08-02:
Got the same:

The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

pibuxd, 2021-04-02:
got same