isDipesh/gnome-shell-extension-sensors #152: Error when install with Ubuntu 20.04

Yossarian0916, 2020-08-03:
Cannot install the extension in my newly-updated OS
got error:
```
The settings of extension temperature@xtranophilist had an error:
SyntaxError: missing ( after for

Stack trace:

_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

here is my OS info:
```
OS: Ubuntu 20.04.1 LTS x86_64 
Host: XPS 15 9560 
Kernel: 4.14.159-0414159-generic 
```

peacheym, 2020-08-03:
I had this exact same issue!  Is it a extension issue or an installer issue?

peacheym, 2020-08-03:
@Yossarian0916 It looks like there may be a potential solution to this issue in the comments of PR #150.  Let me know if you have any success!