isDipesh/gnome-shell-extension-sensors #154: Installing on Ubuntu 20.04 shows syntax error in extensionsService.js:204:33 - missing ( after for

MonteCreasor, 2020-08-29:
SyntaxError: missing ( after for

Stack trace:
  _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:222:19
  start@resource:///org/gnome/gjs/modules/script/package.js:206:5
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17

Vinaum8, 2020-09-01:
me too i have the same problem.

mark-starc, 2020-09-04:
Same error , same distro Ubuntu 20.04 

The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```
`

AlexsanderAR, 2020-10-28:
Same error here:
The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

abdilra7eem, 2020-11-07:
Same problem:

The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:22
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

Machou, 2020-11-13:
same.

`SyntaxError: missing ( after for

Stack trace:
  _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:22
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:222:19
  start@resource:///org/gnome/gjs/modules/script/package.js:206:8
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17`

rfnxinger, 2023-08-21:
same here for GPaste:

SyntaxError: missing ( after for

Stack trace:
  _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:22
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:222:19
  start@resource:///org/gnome/gjs/modules/script/package.js:206:8
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17