isDipesh/gnome-shell-extension-sensors #157: Error on Ubuntu 20.10

ox15, 2021-01-29:
I got the following error on Ubuntu 20.10:
```The settings of extension temperature@xtranophilist had an error:
SyntaxError: missing ( after for

Stack trace:
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:22
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

crlshn, 2021-02-21:
The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:22
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

same here
ubuntu 20.10
❯ sudo apt-get install lm-sensors hddtemp 
❯ sudo sensors-detect

trycrmr, 2021-02-22:
Error is probably because JavaScript files use a deprecated for each loop syntax, but the readme redirects folks to use freon https://extensions.gnome.org/extension/841/freon/ . Had no issues thus far on Ubuntu 20.04 with Freon.

sercansebetci, 2021-03-06:
@crlshn I got the same error while installing the [Sensors](https://extensions.gnome.org/extension/82/cpu-temperature-indicator/) plugin. I tried your solution, but it didn't work. 

[image]

@trycrmr I tried your solution, but it didn't work. The version of the [Freon](https://extensions.gnome.org/extension/841/freon/) plugin listed in 841th is only working.