isDipesh/gnome-shell-extension-sensors #159: There seems to be a syntax error on this extension.

AnishDe12020, 2021-07-07:
Just installed the extension, only to find that there is a syntax error in the code and hence the extension refuses to work. I hope this gets fixed ASAP!!!
Details - 
[image]

TheMcSebi, 2021-07-24:
It's still there :/

manoj-ynos, 2021-09-11:
Facing the same issue.

matilde-t, 2021-11-07:
### Same issue

The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:115:22
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:206:19
start@resource:///org/gnome/gjs/modules/script/package.js:190:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

xezo360hye, 2021-11-16:
Yeah, it' still here, installed and got the error immediately

alaahmet, 2021-11-23:
yeah, still exists :/

The-Seyed, 2022-03-11:
Getting the same error...
Any updates on this?