isDipesh/gnome-shell-extension-sensors #162: dosen't work

MTH96, 2022-02-05:
The settings of extension temperature@xtranophilist had an error:
```
SyntaxError: missing ( after for
```

Stack trace:
```
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:204:33
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

hojjatabdollahi, 2022-03-16:
I'm also seeing the same issue on Manjaro.
Gnome Version: 41.3

michael-lurquin, 2022-04-20:
A fork exists here: https://github.com/UshakovVasilii/gnome-shell-extension-freon